intelligent-agent/Reflash #33: Better log handling

eliasbakken, 2023-04-18:
Right now logs are added by the bash scripts. 
The logs are cleared once a new operation is started. 
It would be good to have the option to see logs from all operations
and give the user the possibility to clear the log manually.

eliasbakken, 2023-04-18:
Thanks for the suggestion. Logrotate is not really solving the problem I think. I think the important issue is to see in the browser what is going on during download, upload, install and backup so that we can catch errors like the one where the partition was not large enough. It's also good to get some information on the filenames and files sizes without necessarily having that displayed to the user when everything is working as it should. I do like the idea or some sort of log levels though, so perhaps we could use a bash tool for that and not reinvent the wheel.

goeland86, 2023-04-18:
I mean... The rest way to do this is to modify whatever bash commands you have to concatenate to a log file (or two, so you can have standard output and error output separate). But I need to look in depth at what you're trying to achieve. I may have some time tomorrow for this.

eliasbakken, 2023-04-18:
It's not a huge deal to implement the changes I'm looking for I think. Right now all log files are truncated on the beginning, 
https://github.com/intelligent-agent/Reflash/blob/f1ef1bdf2081cb301f39682fff147a5e0c4a599d/bin/prod/flash-recore#L5
So that's an easy fix. Then we would need a button in the log component in Vue to clear the log and a server side function that will truncate the file. Perhaps a better implementation is to not return the log as part of every request for progress as we do today, but instead have the log in a separate request.

eliasbakken, 2023-12-15:
This has been implemented and merged in main

closed 2023-12-15 by eliasbakken