intelligent-agent/Reflash #50: Remove root access on Reflash

eliasbakken, 2023-12-10:
Reflash is meant as an intermediary server with the purpose of installing Refactor/Rebuild. 
Still, it should not be necessary to have ssh access enabled for root.

goeland86, 2023-12-10:
I don't remember if there's a non-root user account defined on the image though. Not a bad idea to do it, just don't remember what you setup on the Rebuild barebones.

eliasbakken, 2023-12-10:
There is no non-user on Reflash, so we should add one. Perhaps something more obscure than 'debian'?

goeland86, 2023-12-10:
I mean... Reflash the "image" or Reflash the "tool"? If Reflash the app, then yes. For the image, it can still be the debian user, but we need a way to force-change the password before it begins to run.

obgr, 2023-12-10:
Having Reflash running as a service user is good practice.
I do however have the opinion that this user should be different from the user that is open for ssh.

May I propose two users?
- reflash
- debian

eliasbakken, 2023-12-10:
I agree with this. Which user is the service user and which is the ssh user?

obgr, 2023-12-10:
Service user: reflash
Used to run the service which is called reflash. In case the underlying webserver would be a somewhat standardized tool, i would call the service user nginx, haproxy, podman etc.

Regular user: debian
I would keep it as close as default as possible for this user in order to keep it similar across all rebuild/refactor setups, and hopefully  avoiding confusion.
The default user is named different across distros and live environments, But if you look at AWS as an example, they name the default user after the running distro (debian, centos, ubuntu etc.) which makes the username debian somewhat standard.

goeland86, 2023-12-10:
Standard debian nomenclature for the web-server user is `www-data` usually.

`debian` for the login user makes sense.

eliasbakken, 2023-12-10:
Yeah, maybe www-data is a good user for the web-server. I just read that it is the standard for Debian distros.

obgr, 2023-12-10:
I agree. 
Even if its gunicorn, its still a web server.

eliasbakken, 2023-12-29:
Reopening this, since the build is now happening in this repo

closed 2024-06-03 by eliasbakken