intelligent-agent/Reflash #52: Consider always booting to USB drive of connected.

eliasbakken, 2023-12-10:
An alternative to having an on-board reflash would be to always boot to Reflash as long as the drive is connected. 
This would allow the image itself to not have any software specific to Reflash, but u-boot would always boot the USB drive if it is present. 

This would involve educating the user about this behavior and also prompt the user to remove the USB drive before performing a reboot from Reflash.

It would allow an "elevated security level" via the reflash image that can only be accessed by physically inserting the USB drive. Any operation requiring an elevated security level must then be done via Reflash.

eliasbakken, 2023-12-13:
There is a problem with this approach. Ideally, the program should be running in the browser in order to educate the user to remove the USB drive, but since the USB drive is also mounted on root, the program looses the ability to check if the user has removed the drive. 

One solution would be to not reboot the board, but instead shut it down. This would require the user to first remove the USB and then press the boot button. This is not ideal, because the program looses the ability to check if the USB drive has been removed. 

Another solutions would be to have a system that is completely contained in ram somehow and only have a small partition mounted for downloading images. This might be similar to an initrd or ram based root filesystem. Some more research would have to go into this in order to come up with a working prototype. In that case, the program (Reflash) could do polling on the system and wait for the USB drive to be removed before running reboot.

goeland86, 2023-12-13:
You can check if root is on the USB or eMMC by simply verifying if the root partition is `/dev/sd*` or `/dev/mmcblk*`.... I don't see an issue in not being able to tell which drive is booted.

eliasbakken, 2023-12-13:
Yes, it's not problem to tell what medium the system was booted from. 
The issue is that after installing a new operating system to the eMMC, if the system is rebooted, it will boot back into the USB drive as long as it is present. So it is necessary to instruct the user to remove the USB drive while the system is running.

goeland86, 2023-12-13:
Ah, yeah, ok that's fair.

eliasbakken, 2023-12-15:
There is a tutorial for making an Armbian image into a live boot image. I've not been able to follow it and make it work, it might be a bit outdated, but it should serve as a starting point: https://gist.github.com/probonopd/cf4993ca4ab5bd7757a525f19152a123

eliasbakken, 2024-05-14:
Implemented in v1.0.0

closed 2024-05-14 by eliasbakken