intelligent-agent/Reflash #60: Add hostname and IP address to the screen.

eliasbakken, 2024-05-14:
If there is a Recore on the network, Avahi will discover that and find an alternative hostname. Avahi is implemented using systemd-resolved on the latest Reflash, and the name give seems a bit random. I've seen "recore8.local" and "recore9.local" etc. which is a bit less predictable than what avahi-daemon usually does which is "recore-2.local" and similar. Regardless, it would be good to show what the resulting hostname is. It could be good to also add the IP address as a backup. Not everyone knows about the link local address and how to use it.

goeland86, 2024-05-15:
If klipper screen is running, you can anyway access that through the network panel of the settings.

eliasbakken, 2024-05-15:
This is on Reflash, so KlipperScreen is not running. There is a small custom program running that gives the status, just "downloading", "flashing" etc.

eliasbakken, 2024-06-03:
This has been added, but does not show up, probably since the reflash app starts before the network is up.

eliasbakken, 2024-06-04:
Added a timer that updates the IPs after 30 seconds.

closed 2024-06-04 by eliasbakken